jfedor2/flatbox #1: Hi. will be there a 218mm version? for ender3

neoyagami, 2021-08-08:
Hi.
the new design is great. sadly my ender3 is 8mm too small to print it,
so...

will be there a revison for ender 3 and similar (in size) printers?

jfedor2, 2021-08-08:
Rev2 case is 218mm wide and should fit on an Ender 3 (you might have to disable skirt/brim). It is only compatible with rev2 PCB though.

I'm not currently planning to do a 218mm rev1, but who knows.

closed 2021-08-28 by neoyagami

jfedor2, 2021-08-28:
I actually did a redesign of the rev1 PCB to fit a 218mm wide case. I'm currently waiting for the PCBs to arrive and will post the files once I verify everything works. Should be in a week or so.

jfedor2, 2021-09-02:
I just published rev1.1 with a 218mm wide case. Hopefully it fits on your printer. Good luck.