jfedor2/flatbox #2: Repo history rewrite..?

JonnyHaystack, 2021-08-18:
Hi there, it appears you just rewrote the repo history 2 days ago erasing everything from before? I notice that you changed the tactile buttons to be 5x5x5mm now, with the case being redesigned to have parts that flex and act as button tops. Are there any other changes?

Would have been a lot nicer to preserve history and have commit messages describing what has changed, because I didn't even realise for a while. I just ordered 5 of the rev1 PCB the other day and I also ordered 5x5x8mm tactile switches already, so the new case design won't work with those and the old one seems to be gone now.

Thanks for sharing this project by the way, it's really awesome!

jfedor2, 2021-08-18:
Hi, yes, the repository's history has been rewritten and I'm sorry about that.

It's not that I'm not aware how git works, it was done specifically to stop hosting certain files.

If you're halfway through building the original version and don't want to switch to the new one I'm sure you'll figure something out. There were no other changes aside from the buttons. The PCB files weren't modified.

Again, sorry for the trouble.

closed 2021-08-18 by jfedor2

JonnyHaystack, 2021-08-18:
Thanks for the reply, I understand :)