jfedor2/flatbox #4: Any recomendations for 3d filaments?

neoyagami, 2021-08-24:
Hi, im havinhg some troubles with the buttons.?

pla+ one week of use were enoght to break some
petg is hard to put them (those little thingys that secure them on the switch breaks a lot)

any tips in this regard?

jfedor2, 2021-09-03:
I'm afraid I don't have any silver bullet solutions to this problem.

However I have noticed that I have made the stems (legs) on the buttoncaps a little too long. They are 4mm when they should probably be 3mm long. This means that the caps don't go into the switches fully and when pressed they don't rest on the switch, instead putting more stress on the stems than necessary. I'm not sure how much of a difference this makes in practice, but shortening the stems might help with durability. I will try and update the models after I do some more testing, but in the meantime it's easy to do with side cutters.

When it comes to precise fit, every printer might be calibrated differently so it might be worthwhile to try and scale the models a few percent up or down (possibly using the "XY size compensation" function).

hurik, 2022-09-20:
@neoyagami 
I used ABS for the button caps, but I can't say anything about the durability at the moment.
I had problems with PLA, PETG and Resin. The legs were always not printed very good and I had to use a scalpel to get them to fit. Don't know why it is working so good with ABS ...

hurik, 2022-09-22:
> However I have noticed that I have made the stems (legs) on the buttoncaps a little too long. They are 4mm when they should probably be 3mm long. This means that the caps don't go into the switches [...]

I made some the legs of the buttons 0,5mm shorter, to prevent problems because of not perfect prints. Here are the files: [3d-printed-buttoncaps_3p5mm_legs.zip](https://github.com/jfedor2/flatbox/files/9621085/3d-printed-buttoncaps_3p5mm_legs.zip) (Had to upload them as ZIP, because .stl is not allowed. @jfedor2 I also can make an pull request when you want ...)

Soundtoxin, 2023-02-19:
I haven't broken any caps during use yet (although I have yet to really spend a lot of time playing with the Flatbox), but I have broken 3 caps while removing them from switches. One or both of the stems breaks off in the switch. It seems I cannot safely remove the caps at all once they're on. Also, to get the broken plastic out of the switch, I've been melting the side of the broken keycap to the part of the plastic inside the switch with a soldering iron (Pinecil v2) set to 220C. After 10 seconds or so cooling it should be solid enough to pull out the broken bit of plastic (takes a lot of force). May need to attempt the melting part a couple times.
I made the Flatbox rev5, not sure if the keycaps differed between revisions. Looking at the commit history, I think I already used the updated buttoncaps with the shorter stems (I did my builds this year). I've been using Inland brand PLA+ in various colors and printing on an Ender 3 v2 (I have not tried printing other materials on this printer before).

jfedor2, 2023-03-12:
To anyone having issues with printing the buttoncaps on FDM printers, here's another option, order these from JLCPCB:

https://github.com/TheTrainGoes/GP2040-Projects/blob/main/Open%20Source%20Buttons/Choc%20v1/Flatbox%20JLC%20sprue.stl

Choose 9000R or Black resin. It should be very cheap, like 2 or 3 USD for a whole set.

(Unfortunately you can't combine shipping when ordering PCBs and 3D prints at JLCPCB.)

jfedor2, 2024-12-02:
For anyone still having issues with 3D printed buttoncaps, you can now get correctly sized injection-molded buttoncaps on AliExpress.

For example I've had success with these:

https://www.aliexpress.com/item/1005006980269582.html

Make sure you get the 22.5/28.5mm ones.