jfedor2/flatbox #8: Consider RP2040-Zero variant

JonnyHaystack, 2021-12-23:
I've just seen this tiny Pico alternative https://www.aliexpress.com/item/1005003504006451.html which looks like it should be easy to design around and would allow for a much cheaper non-SMD RP2040 version of the flatbox. It has a variant that comes with pin headers as well, which would make the soldering really easy because if you had holes for all the pins on your PCB the builder would only have to solder on one side. Also it has USB C!

jfedor2, 2022-11-10:
An RP2040-Zero based rev5 is now released!

d75cce5

closed 2022-11-10 by jfedor2

mytokat84, 2022-11-10:
Hello, why you dont use 0.96 inç screen and leds?

TheTrainGoes, 2023-02-21:
@mytokat84 with a 0.96" OLED screen added you would need to adjust the pinout to ensure a good I2C pair.  

With the current setup you would be limited to choosing either OLED or RGB but not both as you would be short one pin.  

There are additional pads on the back of the Zero you could use for this but they are a pain to tap into.

mytokat84, 2023-02-21:
but you can add oled and rgb same time if you cancell L3 and R3 Buttons. Isnt it? L3 and R3 is not necessary to playing any fighting game.. Actually I draw a PCB with oled and I just couldnt add RGB because I dont know which SMD part must be used on rp2040 zero. I researced pico fighting boards rgb outputs on pcb bu I couldnt understand that why there is no connections 2 pieces of 5 volt outputs on 4 pin rgb outputs.

TheTrainGoes, 2023-02-21:
@mytokat84 you can replace L3 and R3 with an OLED if you like.  I would need to check the pins and make sure they can be used as a proper pair for the I2C line.  You would also need to add two resistors on the SDA and SCL lines (I use 4.7k).

For the RGB line you need to to have a 5v logic shifter on the data line (move from 3.3v to 5v).  You will also need a cap and a resistor (100nF and 10k).