jfedor2/flatbox #10: release the working files for the cases?

satorusaka, 2022-01-12:
I'm building a few of rev 1.1 Flatboxes but I've bought usb-c version of the pro micro. it is slightly longer than the original pro micro so it wont fit. im thinking if you can release the working files so i can modify it to fit. thanks!

jfedor2, 2022-01-12:
I added STEP files for the case models. You should be able to work with that.

satorusaka, 2022-01-13:
Alright! Thanks a lot!

closed 2022-01-13 by satorusaka

JonnyHaystack, 2022-02-05:
@satorusaka Did you make that modification successfully yet? When you do, please could you share it? 😄 I also wanted to use some of the USB C pro micros I bought, because I used Elite Cs in the first one I made but those are very expensive.

satorusaka, 2022-02-06:
> @satorusaka Did you make that modification successfully yet? When you do, please could you share it? 😄 I also wanted to use some of the USB C pro micros I bought, because I used Elite Cs in the first one I made but those are very expensive.

yeah I've made a few other adjustments to the case including making it compatible with a USB-C Pro Micro. The screw holes have been changed to use m3x8 screws, I've also changed the holes to fit a round-head hex screw flush to the bottom of the case, because of that the case is thicker at 12mm instead of 10mm.

I've also added a lip around the side walls for it to align it self when I put it together.

Here's the .3mf and .step files.
[Flatbox-rev1.1-case usb-c.zip](https://github.com/jfedor2/flatbox/files/8010622/Flatbox-rev1.1-case.usb-c.zip)

JonnyHaystack, 2022-02-06:
Thanks! Will check it out :smile:

natsumi, 2023-02-18:
@satorusaka Are you able to make the screw hold changes to the rev4 case?  It would be nice if the rev4 was also able to use m3x8 screws and have a flush screw