jfedor2/flatbox #11: Could this be updated to support either Lailh v2/Cherry/Gateron low profile?

CallMeAreks, 2022-02-28:
These switches can use the keycap with the cross-style form. This opens much more options to choose from, like [these](https://es.aliexpress.com/item/1005003327722072.html?spm=a2g0o.detail.1000014.3.71e3b5ba3mhtWH&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40050.266918.0&scm_id=1007.40050.266918.0&scm-url=1007.40050.266918.0&pvid=8e6ef45b-c3c9-4167-b235-4970c6424a87&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40050.266918.0,pvid:8e6ef45b-c3c9-4167-b235-4970c6424a87,tpp_buckets:668%232846%238107%231934&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000026795931939%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B212.84%253B-1%253B-1%2540salePrice%253BMXN%253Brecommend-recommend):

[image]

tdpssn, 2022-03-10:
+1 to this suggestion.

Looking at the specs, the V1's appear to be 13.8mm x 13.8mm, so I'm assuming thats the opening on the current box?

V2's are 13.95mm x 13.95mm so it's not too much of a change, but I'm not sure if I'm reading the specs right, as I've never CAD'd anything.

Specs at bottom of respective pages
V1 http://www.kailh.com/en/Products/Ks/CS/321.html vs
V2 http://www.kailh.com/en/Products/Ks/CS/755.html

CallMeAreks, 2022-03-10:
The size I don't think it's too big of a change but probably  it also requires changing the position of the holes where the hotswap sockets/switches are inserted and soldered.

StdVectorBool, 2022-10-29:
The regular [footprint overlay](https://deskthority.net/wiki/Kailh_PG1353_series) shows the V2 actuator pins are compatible with V1 but:
1. Central pin is larger (5mm diameter)
2. Extra unconnected stabilizer pin (opposite corner from actuators)

[image]

Other libraries support dual V1/V2 footprints by enlarging the central pin and adding the extra stabilizer:
* [V1 / V2 solder](https://github.com/perigoso/keyswitch-kicad-library/blob/main/library/footprints/Switch_Keyboard_Kailh.pretty/SW_Kailh_Choc_V1V2.kicad_mod)
* [V1 / V2 hotswap](https://github.com/perigoso/keyswitch-kicad-library/blob/main/library/footprints/Switch_Keyboard_Hotswap_Kailh.pretty/SW_Hotswap_Kailh_Choc_V1V2.kicad_mod)

but this overlay example shows the new stabilizer unfortunately interferes with the elegant Optional Hotswap design:
[image]

Instead of producing two variants (solder vs hotswap) of a new Rev (for Choc V2), it might be faster to just enlarge that center drill to 5mm and flush cut the new stabilizer (or disassemble the switch and remove it).

StdVectorBool, 2022-11-10:
Will update with photos after my JLCPCB order arrives. :crossed_fingers:

StdVectorBool, 2022-11-15:
Manually removed the 3rd unconnected pin rather than flush cutting:
[image]

Lines up flush with the 5mm center:
[image]
[image]

@jfedor2 would you prefer a PR for the rev4 directory instead of forking? Thinking about committing the KiCAD ZIP extracted so it's clear this is a footprint-only change.

StdVectorBool, 2022-11-15:
Still missing a soldering iron but will post photos of dry-assembly when my buttons arrive.  I didn't modify the 3D printed button caps and ordered the Sinoaracde buttons linked by the OP.

StdVectorBool, 2022-11-18:
Dry assembly, still waiting on a soldering iron:
[image]
[image]

Don't have the V1s button/switch to compare but the Sinoarcade buttons protrude 4-6 mm from the 3D printed case.
[image]
[image]

@jfedor2 Any chance you can provide the source CAD files for the STL? Trying to think about how to raise the surface to recess the Sinoarcade caps while still allowing the tact switches to work.

jfedor2, 2022-11-18:
@StdVectorBool Thank you for the pictures. There's a STEP file of the case in the "extras" folder, let me know if that works for you. I can probably get you the Fusion 360 file as well if that's what you're using.

StdVectorBool, 2022-11-27:
@tdpssn @CallMeAreks 
Updated Gerber file available in my fork:
https://github.com/StdVectorBool/flatbox/tree/master/hardware-rev4

You'll need to remove the 3rd unconnected pin.

SuperPrower, 2022-12-15:
Are those low-profile v2 switches compatible with old ('choc') hotswap sockets?

StdVectorBool, 2022-12-15:
Yes, Choc V1 and Choc V2 use same hotswap sockets.  Only minor difference is PCB footprint for stabilizers.

SuperPrower, 2022-12-16:
Good to know, thanks. Did you have a chance to solder the switches on? I wonder how stable they are without that pin.

StdVectorBool, 2022-12-16:
I stuck with the hotswap sockets and did not solder the switches.  The socket keeps them flat against the PCB so they're stable on 2-axis, but if you intentionally twist the button there is a small amount of play in the 3rd direction.  Time will tell.

SuperPrower, 2022-12-16:
Thanks, that's what I wanted to hear. I've ordered all the components and the boards, will write about my experience once I've built everything

SuperPrower, 2023-01-19:
Okay, it took a few attempts to get a 3d-printed case done, but I have finished my build of @StdVectorBool 's version. 
[image]

I've used FreeCAD to modify the top part of the case from the step file. I've did three things:
1. Raised the walls (and the mounts) by 3mm, so that the buttons won't stick too high above the surface and so that you could do moves like sliding your finger across multiple buttons with comfort and not pop the switch out of sockets. For why this could be useful, see https://www.youtube.com/watch?v=6OULtbPO4fw

This, however, proved to be too much of a raise, at least a little bit. I'm by no means a sophisticated leverless controller user, so I don't know which height level would be comfortable, but I'm going to try and sand the walls back down a little bit over time:
[image]
I imagine that 2-2.5mm additional height would be a sweet spot.

2. Made a cutout in the cylinder parts to fit a brass inserts. I don't like the idea of using a wood screw, since I imagine that the controller would need maintenance, fixes or improvements over time. I've managed to comfortably fit an m2 brass insert and connect the pieces with m2x6mm screws.

3. Since the buttons from Sinoarcade are smaller than the holes, I've reduced the size of their holes slightly - they are now 22mm and 27.8mm for the small buttons and large button respectively. That was just enough tolerance to have almost no visible gaps between the case and the buttons, while still giving enough room to press and tilt the buttons and not have them stuck under the case. I imagine that there is a good scientific way to calculate that tolerance, but it worked out for me, with average quality PETG printing.

There are a few problems that I couldn't solve with my limited CAD knowledge:
1. The printed flap buttons couldn't reach tactile switches anymore, duh. I've ordered a set of tactile switches that are 9.5mm tall from the PCB to the top of the button, this was too tall. I should've measure proper height in the CAD, buuut I just nibbled and sanded them with cutters and files.
2. USB-C port now has a hole on top of it, since I couldn't easily raise it's 'arch' along with the rest of the walls, with how its cutout integrated into the rest of the case:
[image]

So, in the end, I don't think it's just enough to have a modified PCB. There is some (more) work that would need to go into making a case fit a proposed combination of switches and buttons. If anyone is interested, I'm attaching my FreeCAD file, though there is nothing much of interest:
[top.zip](https://github.com/jfedor2/flatbox/files/10452340/top.zip)

SuperPrower, 2023-01-20:
Here is a v2 of the top part of the case. It's a bit more organized for a FreeCAD file, should be easier to edit (hopefully...). Don't know when I'll get to print it and try it out, since I don't own a 3D printer and did the printing at local hackspace, but if anyone wants to give it a try, here are the FreeCAD and stl files:
[top2.zip](https://github.com/jfedor2/flatbox/files/10462079/top2.zip)

Changes:
1. The case is only risen by 2mm instead of 3mm. Don't like the feel of 3mm, and that I can't slide across the buttons without my finger catching the edge of the button hole. And, judging by how other people's flatboxes (with v1 switch) look, I think this should be more or less optimal without buttons sticking out too much. I still have no idea how tall buttons "should" be above the case. Regardless, this can be easily adjusted.
2. Through the magic of the druids, I covered up old USB-C cutout and recreated it a bit higher. Since it's only higher by the same 2mm, it should cover the USB-C just fine, like before, without getting in the way or leaving a hole:
[image]
3. Slightly adjusted the paddings for button holes, increased it not by much (around ~0.05mm), just halved the difference between button radius and the hole radius. Didn't like how close the tolerance turned out with the PETG. Again, should be easy to adjust if needed.

(On that topic, I find the keys somewhat small for my larger fingers. If anyone knows where to buy larger, proper 24 and 30mm keys, on Ali or in Europe, please let me know. Yes, printing them is an option, but I like the quality of manufactured buttons)