jfedor2/flatbox #13: PCB design alternatives?

PutPasswordHere, 2022-03-08:
Don't have much experience with PCB design. I've been checking JLCPCB the past couple months, and a couple parts have been out of stock. Would it be possible to replace some of the parts with alternatives? Would it require a new gerber file?
[image]

jfedor2, 2022-04-11:
If it's a part in a standard package like the resistor (or even the voltage regulator) then you can search JLCPCB inventory for a replacement with the same package and specs. For common parts it shouldn't be a problem and you can use the same Gerber files. If it's the RP2040 or the USB connector that's out of stock then I'm afraid the only option is to wait until it's in stock again.

PutPasswordHere, 2022-04-11:
Appreciate the response. Wasn't sure how lenient some of the parts could be (probably something I could have researched more myself, sorry). Was able to find a time when everything was in stock and snagged them. Thanks for the designs and effort.

closed 2022-11-10 by jfedor2