jfedor2/flatbox #17: Won't Detect or Power After Firmware Flash and Unplug

ktrieun, 2022-05-25:
### Expected Behavior:
After flashing the firmware, I should be able to unplug the controller and have the controller be detected once it is plugged in again.

### Actual Behavior:
It initially detects the controller immediately after flashing the firmware according to the instructions. Once device is unplugged form PC, Flatbox is not detected by computer.

### Steps to Reproduce the Problem:
1. Plug USB into Flatbox PCB and PC
2. Press the RESET switch while holding down the BOOT switch
3. Mount and open "RPI-RP2"
4. Copy "flatbox-rev4.uf2" to "RPI-RP2"
5. Unplug controller from PC
6. Plug controller back into PC

Specifications
Version: Flatbox Rev4
Platform: Windows 10 21H2 & EndeavourOS

jfedor2, 2022-05-25:
Perhaps an issue with the flash chip on your board?

ktrieun, 2022-05-25:
I believe this might be the issue. I've gotten 2 out of the 5 board to work and am waiting on parts to replace.

closed 2022-07-28 by ktrieun