jfedor2/flatbox #18: About the choice of switch

swhan111, 2022-06-09:
hello.
Thanks to your work, I am having fun making controllers.
I have a question about the choice of switch.
Please forgive me for my lack of knowledge about PCB.
1. Can I use Kailh Low Profile Backlight Switch on your PCB?
2. If 1 is possible, does the LED work normally?

Wait for your reply.
thank you

jfedor2, 2022-06-09:
Do you have a link for the switch you have in mind?

In any case, this PCB has no support for LEDs so that's unlikely to work.

swhan111, 2022-06-09:
Thank you so much for your quick reply.

https://www.aliexpress.com/item/4000419542482.html?spm=a2g0o.order_detail.0.0.272c5ccdCXdaj7&gatewayAdapt=glo2kor

I purchased "choc red" from the products in the link above.
And I think this shouldn't be a problem.

However, I have one more question.
First of all, I don't know much about PCBs.
I have nevertheless completed my order with JLCPCB according to your
detailed description, but I have a question about an unresolved issue while
talking to JLCPCB today.

JLCPCB has sent me an email with the following contents.

================
Hi Sir or madam,
Well got your order with many thanks.
Sorry to bother you, but there is one thing that we want to confirm with
you about your SMT order before proceeding.
Since we are not so sure about the polarities of the components are correct
or not. Could you please kindly check if the polarities and placements of
the components are correct in the below picture? Is it okay to proceed with
production?
Your early reply will be highly appreciated, thank you so much!
================

JLCPCB sent the image I attached along with the above e-mail content.
When you check the image sent by JLCPCB, please check whether it is
configured correctly in terms of "polarity" and "placement".

Thanks again for your reply.

jfedor2, 2022-06-09:
The switch should work fine (again, no LED).

As for JLCPCB, I didn't have this issue when I ordered my PCBs, they just fixed the orientations and placement of the components on their side without consulting with me. (It's a necessary step because JLCPCB uses a different convention for their parts library than KiCAD.)

I'm not seeing the image you attached, perhaps post it via GitHub's web interface instead of email.

swhan111, 2022-06-09:
I will upload the image through GitHub's web interface. I ask for your confirmation.
thank yo
[image]
[image]
u

jfedor2, 2022-06-09:
Orientation of U3 looks wrong, pin 1 (red dot) should be top-left.

The rest looks OK I think.

swhan111, 2022-06-10:
Sorry, but I want to ask you one more question.

Are cherry MX switches completely incompatible with current PCB layouts?

If it is not possible to use cherry MX switches in your PCB layout, is it okay if I try to modify your work so that the cherry MX switches are compatible?

thank you

jfedor2, 2022-06-10:
They're incompatible, the footprint of the switch is different.

And sure, go ahead, that's why I provide the files!

closed 2022-06-10 by swhan111