jfedor2/flatbox #20: More of a question

jh0nny2k, 2022-08-22:
How snug is the PCB inside the case? I would like to use your 1.1 design, but with an Arduino pro micro with USB C. This one is 1 mm taller than the micro b connector one.

Do you think it would fit? 

Thanks for the great design files, really love your work.

jfedor2, 2022-08-23:
Vertical space above the (Flatbox) PCB is 4 mm. I'd try it, worst case scenario you will have to make the cutout bigger with a file or something.

jh0nny2k, 2022-08-23:
Thank you. Appreciate the reply. One last question. How slow is the arduino? Is it that slow compared to the other 2? I want to pick this option because it is the cheapest. The board asembly is 45 euros plus 15 euros the boards. Just the boards is 15 euros.

Am i trading that much?

jh0nny2k, 2022-08-23:
i was looking trough your design for the rev 4 :) you must have a big brain:))) you must be really smart. the design is so great. congrats dude, this is some piece of art.

jfedor2, 2022-08-24:
I'm not sure if I ever tested input lag on the Pro Micro version. I'd have to check. In any case, if you want the verified as-good-as-it-gets latency of the RP2040 version, but keep the cost down, you could probably use the rev1.1 PCB with an RP2040-based board that has a Pro Micro-compatible shape. Both Adafruit and SparkFun make one, probably others as well:

https://www.adafruit.com/product/5302

https://www.sparkfun.com/products/18288

I haven't tested this, but I think it should work. You'd probably have to reconfigure the pin numbers in the firmware, but if you use GP2040 you can do that without recompiling anything.

I'm currently considering a variant that would use an RP2040-Zero board (as suggested in #8), which should also be a cheap option.

jh0nny2k, 2022-08-24:
Thank you for the advices. Really appreciate it. I'll give it a try. See how it fits

jh0nny2k, 2022-08-31:
If i may ask, are the buttons 24's and 30 for the jump? I found a shop in Spain selling buttons for kaihl's 11 24's and one 30 for 8 euro. That is mighty cheap. Cheaper than jlcpcb

jfedor2, 2022-08-31:
Yeah, the small ones are 24 and the big one is 30.

Can you post a link to the store?

When ordering on JLCPCB it would probably be better to combine all buttons into one model. I haven't tried but I've seen people do it.

jh0nny2k, 2022-08-31:
https://karboxarcade.com/products/karbox-button-keycaps-set?variant=43003221770479 take a look

jfedor2, 2022-08-31:
As far as I can tell these should work great. Their controller uses the same type of switches.

closed 2022-11-10 by jfedor2