jfedor2/flatbox #21: Just a question

Blkcofkaiju, 2022-09-07:
Just wondering if you'd do a noir style for the button layout? Or would changing the pcb be a hassle?
Sorry if it's a silly question

jfedor2, 2022-09-07:
It's enough of a hassle that I won't do it for you, but all the necessary files are available if you want to attempt it yourself. It would be a matter of dragging some footprints in KiCad to get the layout you want and modifying the case models so that the holes are in the right places.

Blkcofkaiju, 2022-09-07:
Ill give it a go :) thanks for the quick response! 
Appreciate your design here, looks really nifty

Soundtoxin, 2023-02-19:
@Blkcofkaiju 
> Ill give it a go :) thanks for the quick response! Appreciate your design here, looks really nifty

Did you ever make this change? I would be interested for future builds. (I have already built 10 of the current Flatbox rev5)