jfedor2/flatbox #23: flat box rev4 not detected as controller after firmware loaded

SudoKelvi, 2022-09-08:
After loading firmware onto the rev4 flatbox, it simply loops back to opening as RPI-RP2 rather than being detected as a controller. 

So to reproduce: 

1. Connect rev4 via USB
2. Bolt boot + reset 
3. Copy pre-compiled firmware to RPI-RP2 
4. Windows unmounts and remounts RPI-RP2 
5. Disconnect and reconnect simply re-mounts RPI-RP2 rather than being detected as a controller 

Another bizarre thing is that I dont actually need to hold boot and reset, it always just opens RPI-RP2 by default?

jfedor2, 2022-09-09:
Maybe something wrong with the flash chip? Or I guess it would also behave like that if the BOOT button was stuck in the pressed position or shorted somehow, but it seems less likely. Do you have another board to see if it acts the same?

SudoKelvi, 2022-09-09:
I checked the boot and reset buttons and they seem good? Basically did a simple continuity check and I only get a beep when the button was pressed. I'll check other boards and see what might be up.

SudoKelvi, 2022-10-09:
Okay, so it was definitely the board. Trying another and it worked so I'll figure it out later.

If I can pivot to a new problem: using the latest firmware, sw2 doesn't seem to work? Using a continuity tester, the switch works, and when pressed I get a signal from ground to the RP2040. I think the input is not being used in firmware?

jfedor2, 2022-10-10:
> If I can pivot to a new problem: using the latest firmware, sw2 doesn't seem to work? Using a continuity tester, the switch works, and when pressed I get a signal from ground to the RP2040. I think the input is not being used in firmware?

Yeah, I guess I ran out of ideas for what to map that button to so it's unused in the default firmware.

But it's connected to the RP2040 so it's usable if you flash some other firmware.

closed 2022-11-10 by jfedor2