jfedor2/flatbox #25: Flatbox detected, buttons presses not registering

MorroMorro, 2022-10-03:
Hi J,

Windows is detecting it as flatbox rev4. I added the firmware. Put a switch on the board to test.

Was detecting the button press just fine and then suddenly stopped. Now I've tried 3 boards, 5-6 switches, and non register the button presses. It was working then all of a sudden stopped, so I don't think it's a hardware issue. Any ideas I can try to troubleshoot this issue?

jfedor2, 2022-10-04:
Yeah, the UF2 file copying works as you described, that's normal.

Are you testing the buttons in joy.cpl?

Have you tried alternative firmware like GP2040?

MorroMorro, 2022-10-04:
Morning J. Yeah, I've been testing in joy.cpl. Flatbox looks amazing btw.

Will try alternate firmware and report back. For GP2040, hate to bother but I'm terrible at this stuff, do I just drop all the files and src folder from https://github.com/jfedor2/flatbox/tree/master/firmware-rp2040 onto the controller "drive"?

MorroMorro, 2022-10-04:
No issue at all. I'm just bad at applying switches. J is a hero.

closed 2022-10-04 by MorroMorro

jfedor2, 2022-10-04:
Glad to hear you got it working.

For future reference, the alternative GP2040 firmware can be downloaded from here:

https://github.com/FeralAI/GP2040/releases/download/v0.4.3/GP2040-FlatboxRev4_v0.4.3.uf2

And then you flash it the same way.

Let me know when you make the video!