jfedor2/flatbox #26: start button not getting detected rev1.1

jh0nny2k, 2022-10-10:
Hi, 

I have soldered 2 pcb's and on both of them the start buttons is not getting detected. Being that anyway i have to configure these in steam i just skipped the button in the configuration process. The button presses are also not registered on the micro itself. for all other buttons the led flashes for the start button there is no flash.

Thing is i connected the button directly to the pin (ran a wire) and it still didn't register, Wich is weird. Any suggestions?

Side question, can't the micro be coded to be detected as a 360 controller by windows?

Kind regards

jfedor2, 2022-10-12:
I don't really have any ideas. Does this happen on two different Pro Micros? Do the other pins work as they should? Do you have "Pro Micro" selected in the boards menu?

As for emulating a 360 controller, there are firmwares that run on a Pro Micro and support XInput. I haven't tried them, but here's an example: https://github.com/fluffymadness/ATMega32U4-XINPUT-Fightstick

jh0nny2k, 2022-10-12:
/*
 *  Project     Arduino XInput Library
 *  @author     David Madison
 *  @link       github.com/dmadison/ArduinoXInput
 *  @license    MIT - Copyright (c) 2019 David Madison
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 *
 *  Example:      GamepadPins
 *  Description:  Uses all of the available pin inputs to build a 'complete'
 *                Xbox gamepad, with both analog joysticks, both triggers,
 *                and all of the main buttons.
 *
 *                * Joysticks should be your typical 10k dual potentiometers.
 *                  To prevent random values caused by floating inputs,
                    joysticks are disabled by default.
 *                * Triggers can be either analog (pots) or digital (buttons).
 *                  Set the 'TriggerButtons' variable to change between the two.
 *                * Buttons use the internal pull-ups and should be connected
 *                  directly to ground.
 *
 *                These pins are designed around the Leonardo's layout. You
 *                may need to change the pin numbers if you're using a
 *                different board type
 *
 */

#include <XInput.h>

// Setup
const boolean UseLeftJoystick   = false;  // set to true to enable left joystick
const boolean InvertLeftYAxis   = false;  // set to true to use inverted left joy Y

const boolean UseRightJoystick  = false;  // set to true to enable right joystick
const boolean InvertRightYAxis  = false;  // set to true to use inverted right joy Y

const boolean UseTriggerButtons = true;   // set to false if using analog triggers

const int ADC_Max = 1023;  // 10 bit

// Joystick Pins
const int Pin_LeftJoyX  = 0;
const int Pin_LeftJoyY  = 0;
const int Pin_RightJoyX = 1;
const int Pin_RightJoyY = 1;

// Trigger Pins
const int Pin_TriggerL = A2;
const int Pin_TriggerR = A0;

// Button Pins
const int Pin_ButtonA = 10;
const int Pin_ButtonB = 14;
const int Pin_ButtonX = 16;
[... 117 lines cut ...]
		int leftJoyY = analogRead(Pin_LeftJoyY);

		// White lie here... most generic joysticks are typically
		// inverted by default. If the "Invert" variable is false
		// then we'll take the opposite value with 'not' (!).
		boolean invert = !InvertLeftYAxis;

		XInput.setJoystickX(JOY_LEFT, leftJoyX);
		XInput.setJoystickY(JOY_LEFT, leftJoyY, invert);
	}

	// Set right joystick
	if (UseRightJoystick == true) {
		int rightJoyX = analogRead(Pin_RightJoyX);
		int rightJoyY = analogRead(Pin_RightJoyY);

		boolean invert = !InvertRightYAxis;

		XInput.setJoystickX(JOY_RIGHT, rightJoyX);
		XInput.setJoystickY(JOY_RIGHT, rightJoyY, invert);
	}

	// Send control data to the computer
	XInput.send();
}

jh0nny2k, 2022-10-12:
I modified the above to make it work. Till now it works okey, no complaints. i just lost the pin 0 and 1 buttons, not sure what i could possibly map on them. plus not sure how to map the Xbox button as i just modified the pins in the above programing :)