jfedor2/flatbox #27: Question regarding flash chip alternatives

piuzera, 2022-11-13:
Greetings!

I recently ordered a new batch of PCBs and it seems like the flash chip of two of them are having issues. I can flash the software included in the repo and everything works properly but once I unplug and replug, the controller is no longer recognized.

I suspect this is because of a faulty flash chip. Are there any alternatives for the "Winbond Elec W25Q128JVSIQ"? This part is not available in my country and importing it would be very expensive.

Thanks!

jfedor2, 2022-11-13:
I don't want to give a recommendation that I didn't test personally, but in general you need a Quad SPI, XIP, 3.3V part (in the same package, obviously). GigaDevice GD25Q128ES* maybe? It doesn't have to be 16 MB. The Pico only has 2 MB.

But based on your symptoms the flash chip seems to work, but maybe not 100% right. Even immediately after you flash it, it's already running from flash, it's just that it doesn't go through the normal bootloader in that scenario. At least that's how I understand it. You might try recompiling the firmware, changing some parameters. Some more info here:

https://forums.raspberrypi.com/viewtopic.php?p=1874492&sid=cdb260577d9bb4485fe8c3cebe2c94fa#p1874492

piuzera, 2022-11-14:
Thank you for your help! I'll try out different flashing methods to see if there's any change. Hopefully I don't have to order new parts.

jfedor2, 2022-11-18:
Some other folks asked me about the same issue, it seems to be more widespread.

You can try flashing this firmware and see if it helps.

[flatbox-rev4-flashtest.zip](https://github.com/jfedor2/flatbox/files/10043833/flatbox-rev4-flashtest.zip)

piuzera, 2022-11-18:
> Some other folks asked me about the same issue, it seems to be more widespread.
> 
> You can try flashing this firmware and see if it helps.
> 
> [flatbox-rev4-flashtest.zip](https://github.com/jfedor2/flatbox/files/10043833/flatbox-rev4-flashtest.zip)

Indeed, this file is working properly, the flatbox is surviving the power cycle. What kind of changes did you make to it? I'd like to test it on GP2040 too, I customized some settings from it and I'd like to fix it too.

jfedor2, 2022-11-19:
I will push the changes when I have some time, but I just followed advice from that forum thread and compiled it with the following settings in a board file:

```
// On some samples, the xosc can take longer to stabilize than is usual
#ifndef PICO_XOSC_STARTUP_DELAY_MULTIPLIER
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
#endif

// Use slower generic flash access
#define PICO_BOOT_STAGE2_CHOOSE_GENERIC_03H 1

#ifndef PICO_FLASH_SPI_CLKDIV
#define PICO_FLASH_SPI_CLKDIV 4
#endif

#ifndef PICO_FLASH_SIZE_BYTES
#define PICO_FLASH_SIZE_BYTES (2 * 1024 * 1024)
#endif

// All boards have B1 RP2040
#ifndef PICO_RP2040_B0_SUPPORTED
#define PICO_RP2040_B0_SUPPORTED 0
#endif
```

(It's likely one of the first three, or a combination of them.)

jfedor2, 2022-11-23:
After some more testing by other users I now think it's the crystal oscillator part that's problematic, not the flash chip. The latest firmware has a workaround for that (PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64, as mentioned above).

piuzera, 2022-11-29:
Thanks! It now works flawlessly. I'll be closing the issue.

closed 2022-11-29 by piuzera