jfedor2/flatbox #35: how to get to web loader and Switch 2 Rev 5

JBones866, 2023-07-15:
What button is it to hold down to enter the web configurator on this pin out ? 

Also have built 2 separate boards now and it doesnt seem like I get any response from SW2 is this normal, what is the default setup for this switch ? 

Thanks !

jfedor2, 2023-07-15:
The firmware in this repository is not GP2040. There is no webconfig. And yeah, SW2 does nothing by default.

If you want GP2040-CE, flash this firmware and then hold the first button from the left in the top row when plugging in to enter webconfig:

https://github.com/OpenStickCommunity/GP2040-CE/releases/download/v0.7.2/GP2040-CE_0.7.2_FlatboxRev5.uf2

JBones866, 2023-07-15:
Thanks, was going crazy trying to trouble sw2 !