jfedor2/flatbox #36: Improve tactile button layout

Soundtoxin, 2023-07-19:
I think there is room for improvement in how the tactile buttons are laid out. Something to differentiate every other key by touch would help (sorta like how piano keys have some separation by the black keys).
I was thinking something like an added texture or height difference in the case above them (maybe a line or a dome like homing keys on keyboards). It may also help to put some space between the first three and last three on the PCB so you can feel a gap and know where you are.
I've been playing a lot of PC games with my Flatbox rev 5 the last few weeks and the third button from the left is used for the Steam button on the Steam Deck, which is pretty important (accept game invite, join someone's game manually, quit game, etc.). It's very difficult to find quickly and hit blindly. I often second-guess myself and go back and slowly feel out the buttons one by one trying to make sure I'm on the third one.
If anyone has done some sort of mod to theirs to help with telling apart the buttons, I'd be interested in hearing about that as well.  Maybe I could stick something textured on some of my buttons, like a small piece of sandpaper or grip tape.
I'd say besides the keycaps breaking, this is probably the biggest weakness of the Flatbox I've noticed.
I also find that these buttons are a bit hard to press. I was playing Hammerwatch, where the map is on the Select button by default and is something you need to reference pretty often, and I found it somewhat slow/exhausting to keep pressing Select. I did finally just end up shifting my bindings around and finding a spot for the map on one of the main buttons instead as a workaround. I'm not sure if the main source of the difficulty in actuation is from the inner button itself or the way the case flexes to press the button. I'm using Choc Silvers for my main buttons on the Flatbox, but I'm also used to typing on a keyboard with Kailh Box Navy switches which are fairly heavy but still much easier to press than these tactile buttons.

jfedor2, 2023-07-19:
Yeah, that's a valid point.

You can always remap the button so the one you need is easier to find.

There are also versions of the case where the buttons stick out. Check out the OpenStickCommunity Discord.

Soundtoxin, 2023-07-19:
I don't use Discord. Do you know if any of those cases are available publicly elsewhere? GitHub, Thingiverse, Printables, etc.? For now I'll probably see what sort of small mod I can do rather than print a whole new top to try and fix it, but if I ever built another batch of 5 or 10, I'd hopefully have a better solution by then.