jfedor2/flatbox #40: Rev3 PCB Pin Header Sizes Not Listed

Styggy, 2023-08-22:
Hi, I've tried my best to find which pins are appropriate for soldering as the standard pins I'd use for an Arduino don't want to fit into the Rev3 board and it's not listed anywhere in the documentation.

I'm assuming a 2.5mm pitch with a .5mm pin would be small enough but I just want to be sure before I proceed, thank you. The standard size pins fit the brook PCB but not the custom PCB itself.

jfedor2, 2023-08-22:
Pitch is 2.0mm. When you say the pins don't fit the PCB, do you mean that a single pin won't go through the hole? I think the ones I used were 0.5mm.

Styggy, 2023-08-22:
Yeah they won't go through, that's what I needed though, thanks for the quick response.

closed 2023-08-22 by Styggy