jfedor2/flatbox #41: Where is the circular button file

dron-one, 2023-08-29:
I can't find the 3D file of the button, only the shell, I am Xiao Bai, please tell me how to get this. Did you refer to the file again?

marcelo-franceschini, 2023-08-29:
[image]

Did you take a look at the project's root folder?

marcelo-franceschini, 2023-08-29:
Did you take a look at the project's root folder?

[image]

dron-one, 2023-08-29:
谢谢您。但是按钮是两个脚的，我们按钮接口不是圆形十字的吗，接不上

dron-one, 2023-08-29:
Thank you very much. But the button is two feet, our button interface is not round cross, can not connect

marcelo-franceschini, 2023-08-29:
Two feet? 
Our button?
Are you using a translator to convert from an Asian language to English?

dron-one, 2023-08-29:
This button requires a specific mechanical shaft to work。

dron-one, 2023-08-29:
If you want to use a hot-swappable mechanical shaft, which make and model can be used?

marcelo-franceschini, 2023-08-29:
The buttons would be the same model. You would change only the bottom piece.

[image]

dron-one, 2023-08-29:
Got it. Thank you

marcelo-franceschini, 2023-08-29:
> Got it. Thank you

You're welcome! Could you please close this issue? Thanks

dron-one, 2023-08-29:
okThank you

closed 2023-08-29 by dron-one