jfedor2/flatbox #44: Switches don't fit

SquidRadio, 2023-09-09:
In the instructions, one of the parts listed are "12x Kailh low profile (choc v1) switches of your choice", but when I place the switches to the board, the center piece on the switch is too large to fit within the PCB, which prevents the switches from being solderable or fitting properly in the flatbox at all.

I am using the CPG135301D01 v1 switches with the cross-hatch port, rather than the v2 with the two-prong port. I have the rev4 version of the PCBs.

Is there something I am missing? A visual guide would be helpful as well for newcomers to the project.

jfedor2, 2023-09-09:
You have v2 switches.

SquidRadio, 2023-09-09:
And the v2 switches won't work. Makes sense.

Is there a switch that you recommend? I am new to this sort of project and don't want to end up buying the wrong thing again, as the label on Amazon did say v1 switches, and the listing doesn't have a link to any specific set.

closed 2023-09-09 by SquidRadio

JarTheHead, 2023-09-10:
> Is there a switch that you recommend?

I personally use the Choc V1 crystal red switches (normal red is basically the same). I got them from [Aliexpress](https://www.aliexpress.us/item/3256805260407528.html). There doesn't seem to be any Amazon fulfilled ones.