jfedor2/flatbox #45: Case top heat-set inserts don't fit

StellaTerra, 2023-09-12:
The model for the case top has 2.8mm diameter openings for the screws.
[image]
The instructions call for M3 screws, which have a diameter of 3.0mm. The instructions don't yet call for heat-set inserts, but following this video, I thought they would be appropriate: https://youtu.be/mK5Ez4Nguuw?si=9YgIthzJSPQkv2V7&t=352
When the inserts arrived, and I tried to heat-set them into the case top, and it mangled the print pretty bad. Looking back at the video, it looks like there is a wider area within the hole, and then a narrower area below, which is not present in the STL. In doing some research, it looks like they start at 3.5mm outer diameter and get bigger from there. I don't think these will ever fit well in a 2.8mm hole, no matter how well printed. After a while I put a screw into one of the inserts and jammed it in with a soldering iron, but I don't think the holes are going to line up well with the bottom anymore.
Can I get some clarification on the instructions, are you expecting the use of heat-set inserts? Is the M3 a typo, and we're supposed to be using M2 screws? Do you recommend drilling out the screw holes so that the inserts fit better? Is there a possibility the STL was supposed to have printed threading that is missing?

jfedor2, 2023-09-12:
The instructions don't call for heat inserts, there is no typo. Just drive the screw into the plastic. If the hole was 3mm, it wouldn't really hold a 3mm wide screw, would it?

JarTheHead, 2023-09-12:
> The instructions don't yet call for heat-set inserts, but following this video, I thought they would be appropriate: https://youtu.be/mK5Ez4Nguuw?si=9YgIthzJSPQkv2V7&t=352 

The video you shared appears to be using an SGF Bridget case, a fork of the Flatbox that uses M2 Heatset inserts

StellaTerra, 2023-09-12:
Yes, my bad. Thank you for the clarification.

closed 2023-09-12 by StellaTerra