jfedor2/flatbox #54: Firmware not working

ItsReLLiK, 2024-02-19:
So, I downloaded the GP2040-CE firmware for the flatbox rev 5 and none of the keys function how they are supposed to at all. It looks like the pinout is the stock Waveshare RP2040 configuration. Do you have any suggestions or maybe an older working version for download?

jfedor2, 2024-02-19:
You probably flashed the RP2040-Zero firmware first, realized it's not the right one, then flashed the rev5 firmware. Unfortunately the way this works is that the Zero configuration survives the second flash. You can flash the flash_nuke.uf2 file then flash the rev5 firmware or reconfigure the pins manually in webconfig.

ItsReLLiK, 2024-02-19:
This worked, I did flash the other firmware first without even thinking about it.

For anyone else, after you flash nuke it, you have to unplug it and plug it back in or it won't take. I tried doing it without unplugging the board a couple times and was unsuccessful.

closed 2024-02-19 by ItsReLLiK