jfedor2/flatbox #56: JLCPCB CPL Missing data

statikg11, 2024-05-07:
When uploading cpl file on JLCPCb error message below is generating. any solutions?

[image]

jfedor2, 2024-05-07:
These are the keyboard and tact switches, they're not assembled by JLCPCB, so just ignore it and continue.

(They shouldn't be in the BOM file in the first place.)

statikg11, 2024-05-08:
so what files should i drop here
[image]

jfedor2, 2024-05-08:
I just meant I shouldn't have included those entries in that file. Use the same files and just click "continue".

statikg11, 2024-05-08:
Ok so use the cpl file twice?

jfedor2, 2024-05-08:
No, use BOM and CPL files as you did and just click "continue".

statikg11, 2024-05-11:
Thank you. I was able to upload all required files turns out it was an error with the website. Customer support was able to troubleshoot it and resolve.