jfedor2/flatbox #58: KiCad files

laurazard, 2024-06-05:
Hiya,

I'm trying to make some changes to the PCB so I can play around/print my own (and also learn about PCB design), but I've only found the Gerber files. Since (it seems) KiCad is being used for the PCB design, I was wondering if it would be possible to share the KiCad files for the different revisions? I'm particularly interested in the rev. 1 files, but all/either of them would be a great start.

Thanks! (and please let me know if this isn't possible for one reason or another, or if I just missed them 😅).

jfedor2, 2024-06-05:
The KiCad files are in the "extras" folder for each revision.

laurazard, 2024-06-05:
Duh 🤦‍♀️ I think I might've checked that folder for [rev 1](https://github.com/jfedor2/flatbox/tree/master/hardware-rev1/extras), which doesn't include the KiCad project files, just a step file, but I didn't check the others. Thanks!

closed 2024-06-05 by laurazard