jfedor2/flatbox #60: Rev3 soldering help

sashraf16, 2024-06-10:
Hello, 

First off, thanks for creating this project, I'm sure it'll work amazing, right after I have it fully set up. 

I'm having some difficulty with the soldering potion. I have my flatbox PCB, and my P5 mini board, but the headers seem too large to pass through the PCB, and two far spaced to pass through either the P5 or the PCB. Am I approaching this incorrectly? Just using standard 0.1"/2.54mm single row headers. 

Thanks!

sashraf16, 2024-06-10:
I found this [issue](https://github.com/jfedor2/flatbox/issues/24), seems to be similar, but I'm still not sure if I just use strength to force the pin through the PCB and if I have to manually separate the entire row of headers to try and make the gap between them smaller to fit the pattern on the board

jfedor2, 2024-06-10:
I think you need 2mm pitch headers, not 2.54mm.

sashraf16, 2024-06-10:
Oh I see, that would make sense. 

Would something like this be worth trying? 
[Amazon](uxcell a15062900ux0315 40Pin 2 mm Single Row Straight Male Header Connector Strip 18 Pcs https://a.co/d/i4g4bGV)

And to clarify, each hole should be soldered, correct?

jfedor2, 2024-06-10:
Yeah, looks like it.

I think some of the holes are not used, but it won't hurt to solder them too.

sashraf16, 2024-06-10:
Sweet I will try that and update here so it can be closed out, I think that should fix the issue. Will try to give it an attempt in the next couple of days

sashraf16, 2024-06-12:
I think the 2mm worked, I was able to (at least try) solder the boards together. I think we can close this issue, but I was wondering if you knew how to flash the firmware? I guess I hold reset and boot, as per Brook, but not sure which those are. Unless I'm completely off.

jfedor2, 2024-06-12:
I don't know, but surely the board came with some version of the firmware already flashed, right?

sashraf16, 2024-06-12:
I think it's a soldering issue, gonna take some time to sort this out. Thanks for all the help and clarification otherwise!