jfedor2/flatbox #61: Switches not working, Checked with multimeter

R-Bit-Git, 2024-06-16:
Hey Everybody,

I was hoping to get some help from people smarter than me. I'm having issues with some switches not working, namely switches sw2, sw5 ,sw6 (labels from PCB). sw1 worked momentarily but has since stopped.

The PCB connections between the poles were checked with a multi-meter before soldering the switches in, all worked. The connections were checked again after the switches were soldered in, all worked. after Connecting the RP2040-Zero the connection were checked between the switch pole and the RP pole and after pressing the button the signal came through. Some connections didn't work but after rechecking and adding to the solder the connections worked.

When connected to my PC and using the online gamepadtester (Found here https://hardwaretester.com/gamepad) the aforementioned switches didn't work. I wouldn't mind not having the use of them this being my first build and all, I could make do, however missing the sw2 makes it impossible to open the web configurator to rebind the keys from the current odd layout

any ideas or a nudge in the right direction would be greatly appreciated.

Regards

RBG

jfedor2, 2024-06-16:
What firmware did you flash?

You can always get into webconfig by flashing this by the way:

https://github.com/jfedor2/force_webconfig

R-Bit-Git, 2024-06-16:
I flashed the GP2040-CE_0.7.9_WaveshareZero

thanks for the webconfig, will give it a go.

jfedor2, 2024-06-16:
That's not the right firmware, you should have used the Flatbox rev5 one.

Probably easiest to wipe the board by flashing flash_nuke.uf2 and flash the correct firmware.

If you want to go to webconfig with the current firmware, just flash the force_webconfig.uf2 file same way you flashed the firmware (hold BOOT etc.). It will not change the firmware, just go to webconfig once.

R-Bit-Git, 2024-06-16:
Thank you so much that seems to have fixed it, sorry about the rookie error with the firmware, and thanks for the work you've done :D

closed 2024-06-16 by R-Bit-Git