jfedor2/flatbox #64: Flatbox rev5 registering as Hori.Pokken controller

BusFinn, 2024-08-24:
I bought a rev5 flatbox online a year or so ago, it originally registered through steam that it was an Xbox controller but at some point it started registering as a horri.pokken controller. I’ve flashed the controller and done the updates on the controller and have had no luck. I would really like to use this on other consoles other than my PC but I don’t think it’ll work if it keeps registering as the horri.pokken controller.

jfedor2, 2024-08-24:
You can switch modes by holding specific buttons while plugging the stick in or through webconfig.

https://gp2040-ce.info/usage/

BusFinn, 2024-08-24:
Thanks, I’ll try that and get back to you

BusFinn, 2024-08-25:
Yep that worked thanks so much!

closed 2024-08-25 by BusFinn