jfedor2/flatbox #74: Rev 9 Firmware Flashing Additional Instructions

HuddyP, 2025-09-28:
Are there extra steps in preparing the board for flashing the firmware on the Rev 9 that should to be added to the README for newer builders? From what I've tried to research, getting OpenOCD set up for commands to register in CMD to complete the first few steps has been a huge pain point that I cannot get figured out as somebody that's unfamiliar with flashing firmware for these kinds of boards

jfedor2, 2025-09-28:
No, these are all the steps.

ReggaeUlli, 2025-09-28:
I just did it yesterday. After downloading openocd you need to add it to path, so that the command is found. Otherwise you call the openocd exe with its full path. Then i did not get openocd to run as it said it "could not open the device".

the problem for me was that i needed to install drivers for my USB Debug interface.

So i plugged it in via USB and then in file Explorer within this USB device. There was a folder with drivers i manually needed to install. But this will be different for each different brand.

Those are the only two steps i needed which are not in the Dokumentation, but i think AS they are Plattform dependent for Installation and brand dependent for the drivers it is not so helpful to add.